X-Ryl669/eMQTT5 #1: Cannot build with MQTTOnlyBSDSocket set to 0

juhda, 2020-06-04:
Cannot build when setting the value of `MQTTOnlyBSDSocket` to `0` in `lib/include/Network/Clients/MQTT.hpp` because `lib/src/Network/Clients/MQTTClient.cpp` tries to include files that do not exist in the repository.

X-Ryl669, 2020-06-04:
Yes that's true. Those files are part of a very large networking multi-protocol library that we use internally for our needs and it's not open source (yet ?).

You have to define `MQTTOnlyBSDSocket` to `1`. If you need TLS, you'll need to download and use `MbedTLS` library in that case. For an example code, you have [the related repository](https://github.com/X-Ryl669/esp-eMQTT5)

X-Ryl669, 2020-06-04:
The code will build and work correctly with `MQTTOnlyBSDSocket` set to `1`, I meant.

juhda, 2020-06-04:
I understand. Thank you for the clarification.

We have a scenario when our application is compiled with Visual Studio, in which case BSD sockets are not available. That's why I started to look around in the code and tried to change the macro; I hoped it might help.
Is there any straightforward solution to build with Visual Studio?

X-Ryl669, 2020-06-08:
Do you need SSL ?

Currently, I can easily add the missing `Socket.hpp` file but `SSLSocket.hpp` requires so much dependencies (mainly OpenSSL wrappers but also all crypto code that have nothing to do with this project). 

I'm sorry I'm not using Windows anymore since many years. I know the `Socket` class is cross platform, but I don't know for `SSLSocket` status.

juhda, 2020-06-08:
No need for SSL (currently).

juhda, 2020-06-16:
Looks good. Thank you very much!

closed 2020-06-16 by juhda