X-Ryl669/eMQTT5 #2: missing functions getEthernetRate and getWIFIRate

wrmbaron, 2020-06-05:
I am compiling on debian 9 and the linker does not find getEthernetRate() nor getWIFIRate(). I searched the includes and the libraries and the internet and found nothing.

Please where can I get those functions?

(I had to add link libraries pthread and dl to target MQTTc as well and lower the minimum cmake version to 3.8).

X-Ryl669, 2020-06-07:
Sorry, I forgot to commit those files. Meanwhile, you can replace both line 859 and 873 of `Address.cpp` by `int speed = 0;` to remove both symbol requirement.

BTW, you can compile without those files by defining `MQTTOnlyBSDSocket` to 1

X-Ryl669, 2020-06-07:
Should be pushed by now, please try again.

wrmbaron, 2020-06-08:
Thanks for the quick fix. However, I still have to do the following to make it compile:

    diff --git a/CMakeLists.txt b/CMakeLists.txt
    index b4a47cc..7065217 100644
    --- a/CMakeLists.txt
    +++ b/CMakeLists.txt
    @@ -1,4 +1,4 @@
    -cmake_minimum_required(VERSION 3.13)
    +cmake_minimum_required(VERSION 3.9)
    
    # Fix behavior of CMAKE_CXX_STANDARD when targeting macOS.
    if (POLICY CMP0025)
    @@ -11,4 +11,4 @@ option(REDUCED_FOOTPRINT "Whether to enable reduced footprint for the client cod
    
    
    add_subdirectory(lib)
    -add_subdirectory(tests)
    \ No newline at end of file
    +add_subdirectory(tests)
    diff --git a/tests/CMakeLists.txt b/tests/CMakeLists.txt
    index 0009966..71467de 100644
    --- a/tests/CMakeLists.txt
    +++ b/tests/CMakeLists.txt
    @@ -33,4 +33,4 @@ IF(CMAKE_BUILD_TYPE STREQUAL "MinSizeRel")
    ENDIF()
    
    
    -target_link_libraries(MQTTc LINK_PUBLIC eMQTT5)
    +target_link_libraries(MQTTc LINK_PUBLIC eMQTT5 pthread dl)
    diff --git a/tests/ClassPath/src/LinuxSpecific.cpp b/tests/ClassPath/src/LinuxSpecific.cpp
    index d3d85d9..abbbed5 100644
    --- a/tests/ClassPath/src/LinuxSpecific.cpp
    +++ b/tests/ClassPath/src/LinuxSpecific.cpp
    @@ -1,5 +1,6 @@
    +#include "../../lib/include/Types.hpp"
    +
    #if defined(_LINUX)
    -#include "../../include/Types.hpp"
    
        // We have to build this out of the standard header, as we are using kernel stuff that conflicts with userspace include
        #include <sys/ioctl.h>

I guess linking will be a problem like this on windows (probably conditional linking needed), but I cannot check on that at the moment. Feel free to apply any of the above.

X-Ryl669, 2020-06-08:
I have to think about this a bit more. This is clearly linux specific and I'm pretty sure there's a good cmake rule to let it add the right library & flags by itself without having to list `pthread` and `dl` by hand.

X-Ryl669, 2020-06-08:
WBTW, why are you using CMake 3.9 ? It's almost 3 years old by now. What distribution is it ?

wrmbaron, 2020-06-08:
> BTW, why are you using CMake 3.9 ? It's almost 3 years old by now. What distribution is it ?

Debian 9, hopefully moving to Debian 10 soon. The point is, that you are not using features beyond version 3.9, so why make a higher version the minimum requirement?

X-Ryl669, 2020-06-08:
Well, I've used 3.13 because that's the version in `Debian buster`. `Debian stretch` uses 3.7 and it seems it does not like my generator expressions. So, 3.9 did not seem to be a standard pivot version to choose from. I'm trying to see what is the minimal version I can use. 

I'll let you know when I'm done. Thanks for helping!

X-Ryl669, 2020-06-08:
Ok, I've added continuous integration with building on old Debian Stretch. Let me know if it fixes your build on your side. 

Thanks!

wrmbaron, 2020-06-08:
Works perfectly now, thanks for the fast reaction!

closed 2020-06-08 by wrmbaron