X-Ryl669/eMQTT5 #4: There seems no definition of Platform::free and so on for macOS.

shinyaohtani, 2021-02-16:
my clang says "Undefined symbols". Does eMQTT5 support macOS?

```
Undefined symbols for architecture x86_64:
  "Platform::free(void*, bool)", referenced from:
      Network::Client::MQTTv5::connectTo(char const*, unsigned short, bool, unsigned short, bool, char const*, Protocol::MQTT::Common::DynamicBinDataView const*, Protocol::MQTT::V5::WillMessage*, Protocol::MQTT::V5::QualityOfServiceDelivery, bool, Protocol::MQTT::V5::Properties*) in libeMQTT5.a(MQTTClient.cpp.o)
      Network::Client::MQTTv5::Impl::Impl(char const*, Network::Client::MessageReceived*, Protocol::MQTT::Common::DynamicBinDataView const*) in libeMQTT5.a(MQTTClient.cpp.o)
      Protocol::MQTT::Common::DynamicString::readFrom(unsigned char const*, unsigned int) in libeMQTT5.a(MQTTClient.cpp.o)
      Network::Client::MQTTv5::Impl::~Impl() in libeMQTT5.a(MQTTClient.cpp.o)
  "Platform::malloc(unsigned long, bool)", referenced from:
      Network::Client::MQTTv5::connectTo(char const*, unsigned short, bool, unsigned short, bool, char const*, Protocol::MQTT::Common::DynamicBinDataView const*, Protocol::MQTT::V5::WillMessage*, Protocol::MQTT::V5::QualityOfServiceDelivery, bool, Protocol::MQTT::V5::Properties*) in libeMQTT5.a(MQTTClient.cpp.o)
      Network::Client::MQTTv5::Impl::Impl(char const*, Network::Client::MessageReceived*, Protocol::MQTT::Common::DynamicBinDataView const*) in libeMQTT5.a(MQTTClient.cpp.o)
  "Platform::realloc(void*, unsigned long)", referenced from:
      Protocol::MQTT::Common::DynamicString::readFrom(unsigned char const*, unsigned int) in libeMQTT5.a(MQTTClient.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

X-Ryl669, 2021-02-16:
I was developing on a Mac when I wrote this, so it should work. Can you send me your CMake's build configuration?

X-Ryl669, 2021-02-16:
The definition of the 3 methods are in `eMQTT5/tests/ClassPath/src/Posix.cpp`.

X-Ryl669, 2021-02-16:
Can you add a `-D_POSIX` to your CXX flags to see if it works ? I wonder if they removed the preprocessor macro recently.

shinyaohtani, 2021-02-16:
Thank you for very quick reply! Now I'll try `-D_POSIX`.

shinyaohtani, 2021-02-16:
I added `-D_POSIX`. but result was same.

I indicate `eMQTT5` as a link target from my CMakeLists.txt. Is it correct? What is a correct target name? 
Is it written in the document below? I can't read it because URL is collapse.

> API Documentation
> You'll find the client API documentation here.
> https://blog.cyril.by/en/documentation/emqtt5

shinyaohtani, 2021-02-16:
According to my cmake knowledge, it seems that only one source file is compiled for the target `eMQTT5` which doesn't include `eMQTT5/tests/ClassPath/src/Posix.cpp`. 

> add_library(eMQTT5 src/Network/Clients/MQTTClient.cpp)

shinyaohtani, 2021-02-16:
There seems no definition of Platform::free and so on for library `eMQTT5`. Excuse me, could you please tell me the proper way to write my cmake to link eMQTT5 as a library.

now i just write like:
```
TARGET_LINK_LIBRARIES(myProj
  eMQTT5
)
```

X-Ryl669, 2021-02-16:
I've fixed the URL. The library shouldn't need platform code, only the test code does. Let me check...

shinyaohtani, 2021-02-16:
I just use `Network::Client::MQTTv5` as a library. I include `#include <Network/Clients/MQTT.hpp>` from my code.

X-Ryl669, 2021-02-16:
Can you try with latest master ? It should have fixed the issue.

shinyaohtani, 2021-02-16:
I tried but same. Only `src/Network/Clients/MQTTClient.cpp` is compiled for libeMQTT5.a

Is this your modification correct? 
https://github.com/X-Ryl669/eMQTT5/blob/b197dacbaf1aa31df3eb5ef841b6c24744398dcf/lib/CMakeLists.txt#L3

X-Ryl669, 2021-02-16:
Can you try to apply this patch on the latest master and try to build it (it should fail, please post the log):
```
diff --git a/lib/include/Platform/Platform.hpp b/lib/include/Platform/Platform.hpp
index 30d15eb..322e3e8 100644
--- a/lib/include/Platform/Platform.hpp
+++ b/lib/include/Platform/Platform.hpp
@@ -49,28 +49,27 @@ namespace Platform
     /** Get the current process name.
         This does not rely on remembering the argv[0] since this does not exists on Windows.
         This returns the name of executable used to run the process */
-    inline const char * getProcessName() {
+    inline const char * getProcessName()^M
     {
         static char * processName = NULL;
         if (!processName)
         {
-#ifdef _LINUX
+  #ifdef _LINUX^M
             FILE * f = fopen("/proc/self/cmdline", "r");
             if (f) {
                 char buffer[256];
                 processName = strdup(fgets(buffer, 256, f));
                 fclose(f);
             }
-#elif defined(_MAC)
+  #elif defined(_MAC)^M
             processName = strdup(getprogname());
-#else
+  #else^M
             return "program";
-#endif
+  #endif^M
         }
         return processName;
     }
-
-return "this_program"; } // Poor workaround to avoid so many difference between platforms
+#error here^M
 #else
     /** The simple malloc overload.
         If you need to use another allocator, you should define this method
```

shinyaohtani, 2021-02-16:
:pray: Could you push it to a branch?

X-Ryl669, 2021-02-16:
In the latest master, I've removed Posix.cpp file since it was useless (even for tests). Yet, it seems the compiler tries to build it anyway. It's strange.

X-Ryl669, 2021-02-16:
Try the branch called "badStuff", run `rm -rf build && mkdir build` then rerun `ccmake ..` in the build directory and post output. It should fail building.

shinyaohtani, 2021-02-16:
```
[  1%] Building CXX object common_libs/LoadSensor/common_libs/eMQTT5/lib/CMakeFiles/eMQTT5.dir/src/Network/Clients/MQTTClient.cpp.o
cd /Users/shinya.ohtani/unix/cloned/build/ThermoView_cli/common_libs/LoadSensor/common_libs/eMQTT5/lib && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DMQTTOnlyBSDSocket=1 -DMQTTUseTLS=0 -DMinimalFootPrint=1 -D_DEBUG=0 -I~/repos/external/eMQTT5/lib/include -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -mmacosx-version-min=10.15 -std=c++11 -o CMakeFiles/eMQTT5.dir/src/Network/Clients/MQTTClient.cpp.o -c ~/repos/external/eMQTT5/lib/src/Network/Clients/MQTTClient.cpp
In file included from ~/repos/external/eMQTT5/lib/src/Network/Clients/MQTTClient.cpp:2:
In file included from ~/repos/external/eMQTT5/lib/src/Network/Clients/../../../include/Network/Clients/MQTT.hpp:5:
~/repos/external/eMQTT5/lib/src/Network/Clients/../../../include/Network/Clients/MQTTConfig.hpp:89:9: warning: 
      Building eMQTT5 with flags: __Check__BSD [-W#pragma-messages]
#pragma message("Building eMQTT5 with flags: " CONF_AUTH CONF_DUMP CONF_VALID CONF_TLS CONF_SOCKET)
        ^
In file included from ~/repos/external/eMQTT5/lib/src/Network/Clients/MQTTClient.cpp:2:
In file included from ~/repos/external/eMQTT5/lib/src/Network/Clients/../../../include/Network/Clients/MQTT.hpp:8:
In file included from ~/repos/external/eMQTT5/lib/src/Network/Clients/../../../include/Network/Clients/../../Protocol/MQTT/MQTT.hpp:7:
~/repos/external/eMQTT5/lib/src/Network/Clients/../../../include/Network/Clients/../../Protocol/MQTT/../../Platform/Platform.hpp:72:2: error: 
      here
#error here
 ^
1 warning and 1 error generated.
make[2]: *** [common_libs/LoadSensor/common_libs/eMQTT5/lib/CMakeFiles/eMQTT5.dir/src/Network/Clients/MQTTClient.cpp.o] Error 1
make[1]: *** [common_libs/LoadSensor/common_libs/eMQTT5/lib/CMakeFiles/eMQTT5.dir/all] Error 2
make: *** [all] Error 2
```
What does this mean?

shinyaohtani, 2021-02-16:
Oh, link error was gone with master HEAD!! 1 hour ago, I tried it. it seems that it was my mistake. I'm sorry.
https://github.com/X-Ryl669/eMQTT5/issues/4#issuecomment-779793363

Now I can build it without any problems. Thank you.

closed 2021-02-16 by shinyaohtani

X-Ryl669, 2021-02-16:
Ok, great. The issue you experienced was because I was too fast pushing code that did not build. I've fixed up 5mn later but you've probably pulled in between. Glad it works.

commit 603ea783 (2021-02-16): Fix brace error. Fix #4