X-Ryl669/eMQTT5 #7: No support for unsubscribing

yunglook, 2022-05-24:
Why is there no unsubscribe function?

Is there a reason for it? 

If useful, how could it be implemented?

X-Ryl669, 2022-05-24:
I've never used one. Usually, I'm disconnecting from the from the broker (so I'm unsubscribing by default all my subscriptions).
I've added the [unsubscribe method](https://github.com/X-Ryl669/eMQTT5/blob/0e2414eb7c419916125dc9e3f3adac3216c91bca/lib/include/Network/Clients/MQTT.hpp#L205) to the client now, please test it and let me know if it's ok for you.

It's a copy and replace from the subscribe method, I hope I haven't missed anything here.

yunglook, 2022-05-24:
Thanks for the quick response!

You added an unsubscribe method for unsubscribing from a list of topics. What if I wanted to unsubscribe from a single one? 

I added the following function in addition to your changes: 

`
// Unsubscribe from a topic.
    MQTTv5::ErrorType MQTTv5::unsubscribe(const char * _topic, Properties * properties)
    {
        if (_topic == nullptr) 
            return ErrorType::BadParameter;

        // Create the unsubscribe topic here
        Protocol::MQTT::V5::UnsubscribeTopic topic(_topic, true);
        // Then proceed to unsubscribing
        return unsubscribe(topic, properties);
    }
`

However, when I call this function I get an error that returns -1. Did I implement this correctly? Or is it possible to unsubscribe from a single topic with just your changes?

Cheers

X-Ryl669, 2022-05-24:
That's perfectly how you should have written it. Congratulations!

I've added test code for the function and indeed, I was computing the payload size wrongly for unsubscribe packet. The last commit should have fixed the issue. Let me know if it's ok for you!

closed 2022-05-24 by yunglook