X-Ryl669/eMQTT5 #17: readFrom() return BadData for 4-byte buffer

mf01, 2024-01-04:
In [0] there is the check of the variable byte integer buffer length

```
uint32 readFrom(const uint8 * buffer, uint32 bufLength)
{
    for (size = 0; size < 4;)
    {
        if ((uint32)(size+1) > bufLength) return NotEnoughData;
        value[size] = buffer[size];
        if (value[size++] < 0x80) break;
    }
    return size < 4 ? size : (uint32)BadData;
}
```
for `buffer = {0x8c, 0xda, 0xc4, 0x09}` and `bufLength = 4` this will result in `BadData` but it should be a valid input. 
Is the return condition correct?

[0] https://github.com/X-Ryl669/eMQTT5/blob/2f261615d3c2adc5e0c273876cfb35a0c7304b95/lib/include/Protocol/MQTT/MQTT.hpp#L692C31-L692C31

X-Ryl669, 2024-01-04:
You're right, the condition is too restricted. 

I believe the last commit should have fixed it. Please confirm it's ok for you.

mf01, 2024-01-05:
yes, now it seems to be ok. thanks

closed 2024-01-05 by mf01