X-Ryl669/eMQTT5 #19: Adding properties to PUBLISH packet.

turbosapiens, 2024-01-19:
Hi! I tried to pass some props to "publish" method.

`...`

 ```
Protocol::MQTT::V5::registerAllProperties();
Network::Client::MQTTv5::Properties properties;
if (!correlation_data.empty()) {
        Network::Client::MQTTv5::DynamicBinDataView c_data(correlation_data.length(), (const uint8_t*)correlation_data.c_str());
        Protocol::MQTT::V5::Property<Network::Client::MQTTv5::DynamicBinDataView> prop(Protocol::MQTT::V5::CorrelationData, c_data);
        if (!properties.append(&prop))
            printf("PROP APPEND FAILED!\n");
        else
            printf("PROP APPEND SUCCESS!\n");
}
client.publish(publish_topic, (const uint8*) publish_message,
                                            strlen(publish_message), retain_published_message, QoS, (uint16)0U, &properties))
````
                                            
But, i recieve
```
pure virtual method called
terminate called without an active exception
```

in
```
uint32 copyInto(uint8 * buffer) 
{
    o = 1; buffer[0] = header.typeAndFlags;
    o += remLength.copyInto(buffer+o);
    o += fixedVariableHeader.copyInto(buffer+o);
    o += props.copyInto(buffer+o);  //HERE
    o += payload.copyInto(buffer+o);
    return o;
}
```
Maybe i'm just doing something really stupid, but i can't make it work.
But if i just create empty `Network::Client::MQTTv5::Properties properties; ` it works (without any property ofc)

turbosapiens, 2024-01-19:
I also tried your example
```
 // Please do not move the line below as it must outlive the packet
        Protocol::MQTT::V5::Property<uint32> maxProp(Protocol::MQTT::V5::PacketSizeMax, 2048);
        Protocol::MQTT::V5::Property<Protocol::MQTT::Common::DynamicStringPair> userProp(Protocol::MQTT::V5::UserProperty, Protocol::MQTT::Common::DynamicStringPair("key", "value"));
        Protocol::MQTT::V5::ControlPacket<Protocol::MQTT::V5::CONNECT> packet;

        // Check if we have a max packet size property and if not, append one to let the server know our limitation (if any)
        //packet.props.append(&maxProp); // It'll fail silently if it already exists
        packet.props.append(&userProp); // It'll fail silently if it already exists
```
Same behaviour

turbosapiens, 2024-01-19:
Sorry for the inconvenience, i'm idiot
Completely forgot about scope and `if`

X-Ryl669, 2024-01-19:
That's ok, the interface in eMQTT5 is designed for embedded development, so when a pointer is used in the methods signature, it *often* doesn't expect a heap allocated object (or you have to explicitly tell it that it's a pointer to the heap). Dealing with pointers on the stack is not usual for a C++ developer, I can easily understand it's disturbing.

closed 2024-01-19 by turbosapiens