X-Ryl669/eMQTT5 #24: get -2 err code when  connectTo a tcp server with 1883 port

iysheng, 2025-04-11:
why I always get -2 err code when  connectTo a tcp server with 1883 port.
the code with

[image]

X-Ryl669, 2025-04-11:
Error -2 is timeout (see [here](https://github.com/X-Ryl669/eMQTT5/blob/13a984b3e9c26f13635296b1dc9dbb246408a011/lib/include/Network/Clients/MQTT.hpp#L188))

Are you sure there's a server with this IP address and port number listening?

iysheng, 2025-04-11:
> Error -2 is timeout (see [here](https://github.com/X-Ryl669/eMQTT5/blob/13a984b3e9c26f13635296b1dc9dbb246408a011/lib/include/Network/Clients/MQTT.hpp#L188))
> 
> Are you sure there's a server with this IP address and port number listening?

yeah，I also test with the same address and port on esp mqtt client lib，I can connect the mqtt server. 
Besides in another test code I Use addresd format as tcp://a.c.b.d:1883，but in emqtt5 I just use the address as a.c.b.d and port 1883 to call connectTO API，remove the tcp:// prefix。I wonder is this matter ?

X-Ryl669, 2025-04-12:
To answer your question, yes, it's the right way to specify the address (it could be a domain name too, the library is using `getaddrinfo` to extract the underlying IP address). However, the fact that you can connect with esp mqtt library doesn't sound right to me. I've tried to connect to the address you've given here, and I got an AUTH failure:
```
Auth packet received
  AuthMethod:
  AuthData:
  Reason Code: 135
Failed connection to mqtt://47.122.66.243 with error: -6
Bad authentication answer from server
```
Reason Code [135](https://github.com/X-Ryl669/eMQTT5/blob/13a984b3e9c26f13635296b1dc9dbb246408a011/lib/include/Protocol/MQTT/MQTT.hpp#L838C17-L838C30) is **NotAuthorized**.

I wonder if:
1. You've enabled the AUTH feature in the client (you're using ESP32, so you have a menu to enable the features in "components/esp-eMQTT5" when doing `idf.py menuconfig`)
2. Make sure you're using the v2.0.0 branch of the library
3. You've tested locally (on your computer) that it's working. Since eMQTT5 is cross platform, you can build it on your PC with cmake, and then run `(cd build && tests/MQTTc -s 47.122.66.243:1883 -u username -pw somepw -pub /some/topic "Test")` to ensure that it's working locally first.
4. If you have a AUTH scheme required on your broker, you'll need to implement the AUTH logic of your broker in the client. See [this issue](https://github.com/X-Ryl669/esp-eMQTT5/issues/2) for how to do that.

If it doesn't work for you, and to be able to further debug your issue, you'll need to provide some stripped down version of your code so I can try to reproduce here.

iysheng, 2025-04-12:
I debug the code , I found the varible timeout always be 0,

[image]

as you can see although I call ``that()->setTimeout(3000);`` before  I still get that()->getTimeout() equal 0, after I set ``timeout = 3000;`` I can publish msg to server.

I wonder how could this happen ?

iysheng, 2025-04-12:
I make a diff like this 
``` diff
diff --git a/MQTTClient.cpp b/MQTTClient.cpp
index a3c0f82..966e63f 100644
--- a/MQTTClient.cpp
+++ b/MQTTClient.cpp
@@ -475,7 +475,7 @@ namespace Network { namespace Client {

 #if MQTTLowLatency == 1
             // In low latency mode, return as early as possible
-            if (lowLatency && !this->socket->select(true, false, 0)) return -2;
+            if (lowLatency && !that()->socket->select(true, false, 0)) return -2;
 #endif

             // We want to keep track of complete timeout time over multiple operations
@@ -1398,7 +1398,7 @@ namespace Network { namespace Client {
              : ImplBase(clientID, callback, brokerCert, storage), socket(0), timeoutMs({3, 0}) {}
         ~Impl() { delete0(socket); }

-        uint32 getTimeout() const { return 0; }
+        uint32 getTimeout() const { return timeoutMs.tv_sec * 1000 + timeoutMs.tv_usec / 1000; }
         inline void setTimeout(uint32 timeout)
         {
             timeoutMs.tv_sec = (uint32)timeout / 1024; // Avoid division here (compiler should shift the value here), the value is approximative anyway
```
 Now it can work now.

X-Ryl669, 2025-04-13:
Thanks for your investigations. Nice catch, I wonder how I've missed that. I'll reuse your patch in this library first and then port these to esp-eMQTT5 (it's tracking the main library, not the other way). I'll credit you, of course.

X-Ryl669, 2025-04-14:
It's done, in both project, please try the latest master and confirm it works for you, so I can close the issue. Thanks a lot.

iysheng, 2025-04-14:
> It's done, in both project, please try the latest master and confirm it works for you, so I can close the issue. Thanks a lot.

ok, when I build，I get error like:

[image]

after I make diff as
``` diff
diff --git a/MQTTClient.cpp b/MQTTClient.cpp
index fb2c7f0..c917476 100644
--- a/MQTTClient.cpp
+++ b/MQTTClient.cpp
@@ -372,7 +372,7 @@ namespace Network { namespace Client {
     static struct timeval timeoutFromMs(const uint32 timeout)
     {
         return timeval { (uint32)timeout / 1024, // Avoid division here (compiler should shift the value here), the value is approximative anyway
-                         ((uint32)timeout & 1023) * 977};  // Avoid modulo here and make sure it doesn't overflow (since 1023 * 977 < 1000000)
+                         (long int)((uint32)timeout & 1023) * 977};  // Avoid modulo here and make sure it doesn't overflow (since 1023 * 977 < 1000000)

     }
```
I can build success, but I'm not sure this is suite to 64bit support.

X-Ryl669, 2025-04-15:
Ok I think it's fixed. I've also tested different configuration on esp-idf's toolchain, they all build fine. I've added a benchmark comparing with esp_idf's `esp_mqtt` library to show the size difference between both library. Please confirm it works for you, so I can close this issue. Thanks!

iysheng, 2025-04-15:
> Ok I think it's fixed. I've also tested different configuration on esp-idf's toolchain, they all build fine. I've added a benchmark comparing with esp_idf's `esp_mqtt` library to show the size difference between both library. Please confirm it works for you, so I can close this issue. Thanks! 

Ok，Great.

closed 2025-04-15 by iysheng